BBN-Q/Auspex #6: meta info

blakejohnson, 2016-12-21:
- [ ] Convert SegmentNum sweeps to directly accept axis descriptors
- [ ] Construct an experiment from a QGL meta JSON file
    - [ ] Enable filter pipeline downstream from receivers
    - [ ] Update seqfile of AWGs
    - [ ] Construct appropriate `SegmentNum` sweep

blakejohnson, 2016-12-21:
Possible implement bullet item 2 by allowing `QubitExpFactory.create()` to take a `meta_info` argument.

grahamrow, 2017-01-06:
Wen withe the ExpFactory route. Good progress here, though it hasn't been field-tested.

dieris, 2017-01-10:
The seqfiles on the APSs don't seem to be updated

grahamrow, 2017-08-01:
We now expect that metafiles will be run directly with `QubitExpFactory.run(meta_info = ..)` or `QubitExpFactory.create(meta_info = ..)`. SegmentNum sweeps will never be specified manually, since they always derive from the meta_info. Moreover, I've added the method `experiment.add_qubit_sweep` which allows for trivial specification of certain types of sweeps, i.e.:

```python
exp = QubitExpFactory.create(RabiAmp(q, np.linspace(-1, 1, 11)))
exp.add_qubit_sweep("q1 measure frequency")
exp.add_qubit_sweep("q1 measure power")
exp.run_sweeps()
```

closed 2017-08-01 by grahamrow