BBN-Q/Auspex #23: ATS times out when saving single shots with few segments

dieris, 2017-01-11:
A writer directly connected to an Integrator (in order to save single shots) causes the Alazar to fail. This is happening with short sequences (e.g. 2 segments), but not with longer ones (100).

grahamrow, 2017-01-11:
Maybe the writer is getting overwhelmed and the Alazar is timing out? What sort of error output shows up?

dieris, 2017-01-11:
I can't quite figure out what's happening, so a few lines of code are worth more than .... 

Also, if I try to run anything after this, I get the error message `Digitizer ATS2 timed out`

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-9-df5e4e77d962> in <module>()
----> 1 exp.run_sweeps()

c:\users\qlab\documents\github\auspex\src\auspex\experiment.py in run_sweeps(self)
    457 
    458         tasks.append(self.sweep())
--> 459         self.loop.run_until_complete(asyncio.gather(*tasks))
    460         shutdown()
    461 

C:\Users\qlab\Anaconda3\lib\asyncio\base_events.py in run_until_complete(self, future)
    385             raise RuntimeError('Event loop stopped before Future completed.')
    386 
--> 387         return future.result()
    388 
    389     def stop(self):

C:\Users\qlab\Anaconda3\lib\asyncio\futures.py in result(self)
    272             self._tb_logger = None
    273         if self._exception is not None:
--> 274             raise self._exception
    275         return self._result
    276 

C:\Users\qlab\Anaconda3\lib\asyncio\tasks.py in _step(***failed resolving arguments***)
    237                 # We use the `send` method directly, because coroutines
    238                 # don't have `__iter__` and `__next__` methods.
--> 239                 result = coro.send(None)
    240             else:
    241                 result = coro.throw(exc)

c:\users\qlab\documents\github\auspex\src\auspex\filters\io.py in run(self)
    258                     self.data[s.descriptor.data_name, w_idx:w_idx+d.size] = d
    259 
--> 260                 self.file.flush()
    261                 w_idx += message_data[0].size
    262                 self.points_taken = w_idx

h5py\_objects.pyx in h5py._objects.with_phil.wrapper (C:\Minonda\conda-bld\h5py_1474482825505\work\h5py\_objects.c:2705)()

h5py\_objects.pyx in h5py._objects.with_phil.wrapper (C:\Minonda\conda-bld\h5py_1474482825505\work\h5py\_objects.c:2663)()

C:\Users\qlab\Anaconda3\lib\site-packages\h5py\_hl\dataset.py in __setitem__(self, args, val)
    615             mshape_pad = mshape
    616         mspace = h5s.create_simple(mshape_pad, (h5s.UNLIMITED,)*len(mshape_pad))
--> 617         for fspace in selection.broadcast(mshape):
    618             self.id.write(mspace, fspace, val, mtype, dxpl=self._dxpl)
    619 

C:\Users\qlab\Anaconda3\lib\site-packages\h5py\_hl\selections.py in broadcast(self, target_shape)
    295                     tshape.append(t)
    296                 else:
--> 297                     raise TypeError("Can't broadcast %s -> %s" % (target_shape, count))
    298         tshape.reverse()
    299         tshape = tuple(tshape)

TypeError: Can't broadcast (510,) -> (470,)

grahamrow, 2017-01-11:
Looks like the writer is receiving unexpected dimensions somehow. You could try replacing the writer with an `auspex.filters.io.DataBuffer` to eliminate any h5 specific issues.

dieris, 2017-01-12:
it seems that the Averager is receiving an event `DONE` much earlier than the set number of round robins is reached

dieris, 2017-01-12:
Fixed by https://github.com/BBN-Q/Auspex/pull/25

closed 2017-01-12 by dieris