BBN-Q/Auspex #58: XYPlotter doesn't redraw axis as new data are taken

appwagner, 2017-02-02:
XY plotter seems to have problems redrawing the axis when new data are taken. Plots are launched but data doesn't appear. Lots of Bokeh errors: 

C:\Users\qlab\Anaconda3\lib\site-packages\bokeh\models\sources.py:81: BokehUserWarning: ColumnDataSource's columns must be of the same length
  lambda: warnings.warn("ColumnDataSource's columns must be of the same length", BokehUserWarning))

grahamrow, 2017-02-02:
It claims the data dimensions of X and Y don't match, but it would be good to verify this by looking at the data directly. If the code in the `feature/tcexp` branch is up to date, it seems like the descriptors should match. For sanity you might want to create a dummy experiment to spit out data for plot testing?

matthewware, 2019-04-12:
Can we close this issue as Bokeh is no longer in use?

grahamrow, 2019-04-12:
Yes

closed 2019-04-12 by grahamrow