BBN-Q/Auspex #59: Clear plots instead of the full ipython output

dieris, 2017-02-03:
The notebook output is cleared between calibration [runs](https://github.com/BBN-Q/Auspex/blob/6bedcb31ad4878289d5dae93e4b5d7d2ec5a039b/src/auspex/experiment.py#L435). This is good to clean up plots, but it also erases the history of the calibration.

grahamrow, 2017-02-03:
Keeping relevant information? So demanding! I'll have to do a bit more research on this one...

grahamrow, 2017-03-15:
This is actually harder than I thought — getting the current cell output isn't easy.

grahamrow, 2017-05-29:
Plots will now be in a separate matplotlib client. What is the desired behavior if all that remains is text? Just leave it all?

dieris, 2017-05-30:
I think leaving all the text sounds good...we'll try it

grahamrow, 2017-06-07:
No longer and issue with separate matplotlib client.

closed 2017-06-07 by grahamrow