BBN-Q/Auspex #61: run_sweeps no longer consistent with numpy core records

appwagner, 2017-02-09:
Not really sure what happened since I last ran my experiment but after a git rebase origin/master I now get the following error: 

  File "Tc_Measurement_DMM.py", line 347, in main
    cd_exp.run_sweeps()
  File "C:\Users\qlab\Documents\GitHub\auspex\src\auspex\experiment.py", line 481, in run_sweeps
    self.loop.run_until_complete(asyncio.gather(*tasks))
  File "C:\Users\qlab\Anaconda3\lib\asyncio\base_events.py", line 387, in run_until_complete
    return future.result()
  File "C:\Users\qlab\Anaconda3\lib\asyncio\futures.py", line 274, in result
    raise self._exception
  File "C:\Users\qlab\Anaconda3\lib\asyncio\tasks.py", line 239, in _step
    result = coro.send(None)
  File "C:\Users\qlab\Documents\GitHub\auspex\src\auspex\filters\io.py", line 105, in run
    if not np.all(s.descriptor.tuples() == streams[0].descriptor.tuples()):
  File "C:\Users\qlab\Documents\GitHub\auspex\src\auspex\stream.py", line 237, in tuples
    return np.core.records.fromrecords(flattened_list, dtype=self.axis_data_type(with_metadata=True))
  File "C:\Users\qlab\Anaconda3\lib\site-packages\numpy\core\records.py", line 658, in fromrecords
    nfields = len(recList[0])
IndexError: list index out of range

Looks like something in numpy is unhappy.

grahamrow, 2017-02-09:
If you stick a debug before line 105 of io.py and look at the values of stream.descriptor.tuples() for each stream in streams, I'm curious if one of them is malformed. Otherwise it looks like numpy is being asked to do something nonsensical.

appwagner, 2017-02-09:
The problem was mostly on my end. I didn't understand how python treats modifying global variables. The result was that in a statement like: 

my_ouput_var.add_axis(DataAxis("MyLabel",MY_LIST))

MY_LIST was being read as the default [] instead of the modified [a,b,c,d] and so had length 0. The error may have been more obvious if either DataAxis or add_axis complained in the length 0 case. Might be useful to add an exemption there since the add_axis method could start seeing more use.

closed 2017-02-09 by appwagner