BBN-Q/Auspex #70: Data axis points and original_points

dieris, 2017-04-06:
`updating_descriptors` modifies `axis.points` but not `axis.original_points`. [Example](https://github.com/BBN-Q/Auspex/blob/8e24d37788b96ba017945c5cea5d2c257d20499a/src/auspex/filters/channelizer.py#L140). As a result, [`axis.num_points`](https://github.com/BBN-Q/Auspex/blob/8e24d37788b96ba017945c5cea5d2c257d20499a/src/auspex/stream.py#L78) will return the number of `original_points`, even if `points` have changed. This is relevant e.g. [here](https://github.com/BBN-Q/Auspex/blob/8e24d37788b96ba017945c5cea5d2c257d20499a/src/auspex/filters/integrator.py#L44).

As I'm not sure how this interacts with adaptive sweeps, I don't know which one to fix...

grahamrow, 2017-04-11:
Probably update_descriptors should be responsible for setting both. The idea of `original_points` is that the axis can be reset to those original values after any sort of adaptive activity occurs. Otherwise you might have some crazily refined data points being taken in a region with no interesting behavior.

grahamrow, 2017-04-11:
Channelizer fixed in 2b3ecbf, let me know if it's fine.

dieris, 2017-04-21:
yes, that works, thanks!

closed 2017-04-21 by dieris