BBN-Q/Auspex #71: Network analyzer reaveraging broken

gribeill, 2017-04-06:
Driver does not block when reaverage command issued.

caryan, 2017-04-06:
Do you know why it seems to claim it does [here](https://github.com/BBN-Q/Auspex/blob/master/src/auspex/instruments/agilent.py#L290)

gribeill, 2017-04-06:
Yeah, it's pretty weird -- it seems to never hit the loop. I think that there is a slight delay between issuing the averaging restart command and the NA indicating that averaging is not complete. The right way to do this is using the `OPC` GPIB code, so I am implementing that now.

caryan, 2017-04-06:
I guess an alternative is to use `*WAI` like we in the [Matlab version](https://github.com/BBN-Q/Qlab/blob/develop/common/%2BdeviceDrivers/AgilentE8363C.m#L81)

gribeill, 2017-04-06:
Yes, after doing a bit of reading seems that is the friendlier way of interacting through SCPI. Ideally we'd use the [`*SRQ?` event registers](http://www.keysight.com/main/editorial.jspx?cc=US&lc=eng&ckey=602664-1-eng&id=602664-1-eng), but that's going to require asynchronous instruments with event handlers, which is not really a can of worms I want to open right now...

PR #72 (2017-04-06): Fix network analyzer reaveraging.
Fixes #71.
closed 2017-04-06 by gribeill