BBN-Q/Auspex #73: Axis names must be unique

dieris, 2017-04-06:
Currently, the axis descriptor name for e.g. a T1 sequence in QGL is `time`, the same as a `StreamSelector`. This causes [this](https://github.com/BBN-Q/Auspex/blob/8e24d37788b96ba017945c5cea5d2c257d20499a/src/auspex/stream.py#L293) (at least) to break. It seems that `StreamSelector`s should have some protected name.

grahamrow, 2017-04-10:
Hmm, interesting. We can think of some alternative name in Auspex, though  QGL could perhaps refer to this as "delay" since that's typically what is being changed.

dieris, 2017-04-10:
yeah, that was exactly my quick fix, if everyone is okay with it (but at least there should be a warning)

blakejohnson, 2017-04-11:
Renaming the T1/Ramsey/Echo axis to "delay" is fine with me.

dieris, 2017-04-11:
Fixed by https://github.com/BBN-Q/QGL/pull/100

closed 2017-04-11 by dieris