BBN-Q/Auspex #91: AWG Parameter Sweeps

grahamrow, 2017-05-12:
Auspex doesn't currently support the "AWG" sweeps. Add a new parser to exp_factory.py to parse these and add them correctly.

grahamrow, 2017-08-01:
With the new `exp.add_qubit_sweep` we have a pretty good template for how these sweeps should be treated. @dieris @gribeill @matthewware what do you think about:

```python
exp.add_awg_sweep("q1 measure 12:1,2 amplitude") # Quad channel 12 - phys channels 1&2 amp
exp.add_awg_sweep("q1 control 12:1 offset") # Quad channel 12 - phys channel 1 offset
```
It's a little ugly maybe, but it corresponds to drilling some from the qubit yaml to the instrument yaml, and it relatively unfettered by any particular choice of instrument.

gribeill, 2017-08-01:
I think this makes sense. I wonder if there's a way to make custom tab completions...

dieris, 2017-08-09:
I missed this thread. The amplitude sweep should be taken care by https://github.com/BBN-Q/Auspex/pull/113, with the caveat https://github.com/BBN-Q/Auspex/issues/118. 

Sweeping offsets is in my opinion not too common, so we may be fine with calling out the instrument channel explicitly.

grahamrow, 2019-04-12:
Closing since this does not appear to be a pressing issue.

closed 2019-04-12 by grahamrow