BBN-Q/Auspex #103: Prevent multiple plot windows

dieris, 2017-06-30:
Refresh same window for new plots. Or close the window at the start of a new matplotlib session. 

Windows can get really get out of hand in an overnight repeated experiment :)

grahamrow, 2017-07-10:
Okay, let's think of the best way to implement this. I'd like feedback from @gribeill and @matthewware 
The way that the plotting setup works is:
1. Auspex starts a server on port 7771 to serve a JSON plot description to any clients that connect
2. Auspex starts a server on port 7772 for serve data to any plot clients
3. Auspex launches the plot client as a subprocess, with a command line argument to have it connect to localhost.
4. The client obtains the JSON, creates the plots, and then waits for data.
5. When done, auspex sends a "done" signal to the client, causing it to disconnect.

Either:
1. The plotter can continue polling on 7771 after being disconnected (with some timeout) until it finds some JSON descriptor. But, we would need to check to see if there are any active plotters out there (this could be difficult).
2. The plotter can automatically quit after being disconnected as suggested by @dieris 
3. The user should ask that plotters not be connected, and instead launch the client manually and have it keep trying to connect on 7771 whenever it gets disconnected.

gribeill, 2017-08-01:
I think I'd like a slight modification of (2) above... the plotter should persist after a sweep is done, but starting a new process should close the old plotter window and start a new one. Probably the easiest way to do this is just to kill the old subprocess and start a new plot client.

commit d3280363 (2017-08-01): Plotter now gets closed when running exp_factory or when globals.single_plotter_mode is set. Fixes #103
closed 2017-08-01 by grahamrow