BBN-Q/Auspex #104: API Decisions

grahamrow, 2017-07-11:
@dieris @matthewware @gribeill 
Can we keep some running commentary on API changes? Commentary can include "these are terrible ideas"...

- [ ] Implement YAML loading (mostly complete)
   - [X] Eliminate manual channel specification for QGL
   - [X] Convert QubitExpFactory to loading YAML
   - [ ] Derive YAML instrument templates from Instrument classes
   - [ ] Only require YAML config to specify NEW information not in template.
- [X] Run QubitExpFactory directly on a QGL sequence using meta_info:
```python
q = QubitFactory("q1")
exp = QubitExpFactory.run(RabiAmp(q, np.linspace(-1, 1, 21)))
```
- [X] Move primary sweep interface to Python, away from config files, e.g.:
```python
exp = QubitExpFactory.create(PulsedSpec(q))
exp.add_qubit_sweep("q1 measure frequency", np.linspace(6e9, 6.5e9, 500))
exp.add_qubit_sweep("q1 control power", np.linspace(-30, -20, 21))
exp.run_sweeps()
```

dieris, 2017-07-11:
Alright, then I'll start :) These all sound like nice features to me. The main concern I have, though, is user-friendliness. The current API may not be the most efficient way to run loops (especially for more experiments back to back), but I think it's fairly intuitive. I expect that a new, or occasional user, will be more likely to find sweeps and instruments in a GUI rather than somewhere in one or more config file. 

To summarize my comment: 1) I think these inline coding and use of config files is great for experienced users; 2) I suspect that the GUI is still nice to learn (consistent layout, field validation, no need to learn all the function names...) and to figure out what's going on in an experiment quickly. 

I guess what I am saying is that it would be nice to have both functionalities. Just my opinion!

gribeill, 2017-07-11:
After chatting a bit with @grahamrow today, I'm going to try and cook up a little  pyqt GUI that gives you a tree view of the settings and instruments YAML files. This will basically just be a nicer interface to the text version -- that does *no* behind the scenes magic like PyQLab does right now. Later on we can implement an interface that lets you (for example) drag-and-drop templates that are prepopulated with reasonable defaults. @dieris, does this sound like a reasonable compromise?

dieris, 2017-07-12:
sure, I think that would be helpful

grahamrow, 2018-01-23:
We can revisit some of these things later.

closed 2018-01-23 by grahamrow