BBN-Q/Auspex #112: Units for instrument/qubit settings

dieris, 2017-08-07:
It would be useful to have units for instrument settings like frequency, power, ... somewhere in the yml. Both to guide the user (e.g. deg/rad confusion) and to provide the right unit for sweeps.

grahamrow, 2017-08-11:
Hmm. I agree with the notion, but am worried about implementation details. Others have clearly thought about this.
https://stackoverflow.com/questions/20793260/how-to-represent-value-units-in-yaml

I'm a strong proponent of base units, i.e. Hz only, but admit this isn't relevant when discussing  powers and phases. Instrument and filter YAML templates will contain this information and will make their way into the docs, but if we can come up with a good inline solution I'm all for it.

dieris, 2017-08-11:
I see, there is no consensus there. How about templating everything that has a unit as `parameter (unit): value`. Like `frequency (Hz): 6e9`. The value is still at the end, so editing it is not hindered by the unit. The brackets will tell if there is a unit or not.

dieris, 2019-12-12:
YAML has been abandoned

closed 2019-12-12 by dieris