BBN-Q/Auspex #114: Save settings as YAML

gribeill, 2017-08-08:
Settings in `WriteToHDF5` are saved as JSON as opposed to YAML.

matthewware, 2017-08-08:
@grahamrow @gribeill from what I can tell WritetoHDF5 just copies json files to a new folder?  Was there ever functionality to write the setting to h5 files?

dieris, 2017-08-08:
No, it's always been in json. I think we still want to save as yml (now), so that we can easily recover/reload past settings.

matthewware, 2017-08-08:
Okay, I was thinking moving forward we could just do both.  Based on discussions with @grahamrow I'm just going to implement saving to the h5 and to a file.

dieris, 2017-08-09:
That's even better

grahamrow, 2017-08-11:
@matthewware take a look in `src/auspex/config.py` for YAML dumping inspiration. First thing to try might be writing a dumper (a la `yaml_dump`) that doesn't bother calling `add_representer` for `Include` objects? Depends on how flexible the dumper is. Next I'd try modifying what the `Dumper` returns.

dieris, 2017-09-27:
The h5 saving is working. But the yml copies are now limited to the main config file (`measure.yml`). Should we search the file for an `!include` string and load those files? Or can we reuse the `yaml_dump` code to find those files?

dieris, 2017-10-17:
Closed by #134 . It can be improved by saving the settings as the original yml files for better lookup.

closed 2017-10-17 by dieris