BBN-Q/Auspex #116: YAML-compatible calibrations

gribeill, 2017-08-08:
Need to switch over all the calibrations to YAML:

- [ ] Cavity Search
- [ ] Qubit Search
- [ ] Rabi Amplitude
- [ ] Ramsey
- [ ] Phase Estimation
- [ ] DRAG / DRAG2
- [ ] CR

Slowly migrating things over on [`fix/yaml-calibrations`](https://github.com/BBN-Q/Auspex/tree/fix/yaml-calibrations)

gribeill, 2017-08-09:
@dieris, Rabi, Ramsey and Phase Estimation seem to be working for me... can you give them a shot?

dieris, 2017-08-30:
Agree, and merged after a few changes. Keeping this issue open while working through the remaining calibrations.

gribeill, 2017-09-05:
Getting an error now when trying to run calibrations. Will investigate...

grahamrow, 2018-09-14:
What we have has been working, but I'm going to close this since we're moving on from YAML in the near future

closed 2018-09-14 by grahamrow