BBN-Q/Auspex #120: APS and digitizer fall out of sync in 2D sweeps

dieris, 2017-08-10:
Occasionally (again, I hope to circumstantiate this more...), the segment number slips during a repeated sweep (2D). As a result, some of the rows in the 2D data are horizontally shifted. It looks like some triggers are missed at the beginning of these rows, before the digitizer (X6) starts acquiring.

grahamrow, 2017-08-11:
Can you try a delay in exp factory run method a la
```python
        for dig in self.digitizers:
            dig.acquire()
        await asyncio.sleep(0.5)
        if not self.cw_mode:
            for awg in self.awgs:
                awg.run()
```
and see if that makes any difference?

dieris, 2017-08-31:
It definitely helps, although it's unclear what the minimum delay is. Closed by https://github.com/BBN-Q/Auspex/commit/bbae63bff087bffcb9e2d76ef9898b6d65983bcb until further notice.

dieris, 2017-09-24:
Not enough... Trying with a longer delay, but need some deterministic way to figure out how long to pause

grahamrow, 2017-09-25:
Should we explicitly ask the digitizers if they have managed to get themselves ready yet?

dieris, 2017-09-25:
That may be the only reliable way, if not too slow.

dieris, 2017-09-25:
I must also say that with a delay of 0.75 s, I haven't seen any problems yet.

grahamrow, 2017-09-28:
That's pretty painful, we should look into the root cause.

grahamrow, 2019-04-09:
Has this popped up again? @sfallek1 @dieris @matthewware @gribeill

dieris, 2019-06-02:
I haven't seen this happening in a while

closed 2019-06-02 by dieris