BBN-Q/Auspex #128: ZMQ error: address in use

dieris, 2017-08-23:
Frequently running into this error when running a `SingleShotFidelityExperiment`

```
Exception in thread Thread-10:
Traceback (most recent call last):
  File "C:\Users\qlab_user\Anaconda3\envs\pyqt5\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "c:\users\qlab_user\documents\github\auspex\src\auspex\plotting.py", line 78, in run
    self.status_sock.bind("tcp://*:%s" % self.status_port)
  File "zmq/backend/cython/socket.pyx", line 495, in zmq.backend.cython.socket.Socket.bind (zmq\backend\cython\socket.c:5653)
  File "zmq/backend/cython/checkrc.pxd", line 25, in zmq.backend.cython.checkrc._check_rc (zmq\backend\cython\socket.c:10014)
zmq.error.ZMQError: Address in use
```

dieris, 2017-08-24:
Or this ☹️ 
```
ZMQError                                  Traceback (most recent call last)
<ipython-input-6-df5e4e77d962> in <module>()
----> 1 exp.run_sweeps()

c:\users\qlab_user\documents\github\auspex\src\auspex\single_shot_fidelity.py in run_sweeps(self)
     78             self._update_histogram_plots()
     79 
---> 80         self.plot_server.stop()
     81 
     82     def _update_histogram_plots(self):

c:\users\qlab_user\documents\github\auspex\src\auspex\plotting.py in stop(self)
     65             task.cancel()
     66             try:
---> 67                 self._loop.run_until_complete(task)
     68             except asyncio.CancelledError:
     69                 pass

C:\Users\qlab_user\Anaconda3\envs\pyqt5\lib\asyncio\base_events.py in run_until_complete(self, future)
    465             raise RuntimeError('Event loop stopped before Future completed.')
    466 
--> 467         return future.result()
    468 
    469     def stop(self):

c:\users\qlab_user\documents\github\auspex\src\auspex\plotting.py in poll_sockets(self)
     34     async def poll_sockets(self):
     35         while not self.stopped:
---> 36             evts = dict(await self.poller.poll(50))
     37             if self.status_sock in evts and evts[self.status_sock] == zmq.POLLIN:
     38                 ident, msg = await self.status_sock.recv_multipart()

C:\Users\qlab_user\Anaconda3\envs\pyqt5\lib\site-packages\zmq\eventloop\future.py in on_poll_ready(f)
     93             else:
     94                 try:
---> 95                     result = super(_AsyncPoller, self).poll(0)
     96                 except Exception as e:
     97                     future.set_exception(e)

C:\Users\qlab_user\Anaconda3\envs\pyqt5\lib\site-packages\zmq\sugar\poll.py in poll(self, timeout)
     97         elif isinstance(timeout, float):
     98             timeout = int(timeout)
---> 99         return zmq_poll(self.sockets, timeout=timeout)
    100 
    101 

zmq/backend/cython/_poll.pyx in zmq.backend.cython._poll.zmq_poll (zmq\backend\cython\_poll.c:2186)()

zmq/backend/cython/_poll.pyx in zmq.backend.cython._poll.zmq_poll (zmq\backend\cython\_poll.c:2036)()

C:\Users\qlab_user\Anaconda3\envs\pyqt5\lib\site-packages\zmq\backend\cython\checkrc.pxd in zmq.backend.cython.checkrc._check_rc (zmq\backend\cython\_poll.c:2811)()

ZMQError: Unknown error
```

dieris, 2017-08-29:
The plot server similarly wants to stop earlier during a `PulseCalibration.calibrate`

grahamrow, 2017-09-08:
This doesn't really happen to me for non single shot measurements...

dieris, 2017-09-22:
Yeah, this happens when we try to run 2 or more of the following species consecutively: normal runs (QubitExpFactory), single-shot, and mixer calibrations.

dieris, 2017-10-11:
It seems to be a ipynb-only issue

closed 2019-06-07 by dieris