BBN-Q/Auspex #133: Lab Brick power set to -38 dBm!

dieris, 2017-08-29:
Occasionally, the labbrick power is set to this value...

dieris, 2017-09-21:
A more typical behavior now causes the power to be set at -40 dBm, which is the minimum. I wonder if that's the default value when setting the desired value fails.

grahamrow, 2017-09-21:
Could you load the instrument separately and run a power loop for set/get? i.e.
```python
from auspex.instruments import LabBrick
lb = LabBrick("XXX")
for pow in np.linspace(-50,10,1):
    lb.power = pow # set
    print(pow, lb.power) # get
```
to see where it breaks down?

dieris, 2017-09-21:
Good idea, but I ran this loop a few times and nothing breaks down yet...

grahamrow, 2017-09-28:
We discussed putting a piece of verifying code that check to see if the amplitude was set as expected. Could also check after every command to see if the power has been changed as a side effect 🙄.

dieris, 2018-01-30:
This is most likely a driver issue. Getting a similar problem with 
https://github.com/BBN-Q/Auspex/blob/ce75704d4846be2512329bbbf7a174c639cdab34/src/auspex/instruments/vaunix.py#L84
occasionally returning numbers like 18.35 GHz ?!?

matthewware, 2019-04-12:
Have we seen this behavior in the last year?  Something we should try to get to the bottom of?

dieris, 2019-07-31:
This may have been solved here: https://github.com/BBN-Q/Auspex/pull/338. Closing tentatively

closed 2019-07-31 by dieris