BBN-Q/Auspex #148: Frozen attributes of instruments

dieris, 2017-10-06:
@grahamrow I'm confused by how this [freezing](https://github.com/BBN-Q/Auspex/commit/c6f377a83b0bd4a4e50d99841ecffe1d5c5e2c48) works, and by its purpose. Currently, experiments are first loaded (instantiated), then connected, then initialized (`set_all`). Before the `set_all`, the instruments only have a few attributes, like `name`, `address`... At this point all attributes are supposedly frozen and it would be impossible to set parameters like `frequency`, `power`... because they don't exist yet. 
So:
1) Should we instantiate all the attributes at `init`, with `None` values?
2) Is there a reason why the 'frozen' treatment is currently reserved only to `SCPIInstrument`s? https://github.com/BBN-Q/Auspex/blob/5805cd6bdaa68ba148f979f2bfd56425a855a071/src/auspex/instruments/instrument.py#L265-L268

grahamrow, 2017-11-07:
The `Command` objects start life as class variables but are metaprogrammed by the `MetaInstrument` and added as instance variables. Thus, they are available when something like `source.frequency = 10` is run. The goal is to avoid the user calling some erroneous property like `source.fraquncy = 10` which would succeed without warning.

There's no reason we can't do this with non-SCPI instruments, but those drivers tend to call out to other libraries and aren't metaprogrammed in the same way. Thus, we would have to supply a manual list of settable attributes.

dieris, 2017-11-08:
I see, thanks for the explanation. But something must go wrong, because after [connecting](https://github.com/BBN-Q/Auspex/blob/c1f092b536224a142ea07ea6c4f7162b4edf49c8/src/auspex/instruments/instrument.py#L257) the instrument is frozen, and at this point the only attributes for my SCPI instruments are `name` and `resource_name`. A later `set_all` will fail because there are no other attributes. Looking into this...