BBN-Q/Auspex #153: Enable only X6 streams in use

dieris, 2017-10-16:
Sweeps like cavity frequency etc. currently have an overhead of ~ x10 with an X6. Probably due to all streams being enabled.

dieris, 2017-10-16:
Other channels (not in meta-info) are not explicitly [enabled](https://github.com/BBN-Q/Auspex/blob/e5b1ab7226a8ae11a1cfa5a442e6dfb480ecf72e/src/auspex/instruments/X6.py#L147-L148). Even when disabling every other channel explicitly, the overhead remains. Need to profile.

dieris, 2017-10-16:
It turns out that most of the overhead is due to `select` in `pyqt5.lib.selectors`. This function is called ~5k times in a 10-points sweep, for a total of ~7 s.

grahamrow, 2017-11-07:
I'm starting to look into this but am confused as to how pyqt5 is entering into this situation.

grahamrow, 2018-01-23:
@dieris any more recent news here? Could you try the multiprocessing branches to see if things change?

matthewware, 2019-04-12:
I think these issues are fixed now with MP and the new X6 builds?  No?

dieris, 2019-04-14:
yeah, I think we can call this issue obsolete

closed 2019-04-14 by dieris