BBN-Q/Auspex #158: Auspex can't run one segment sweeps

matthewware, 2017-10-27:
Auspex has no notion of running a single segment experiment:

```
from QGL import *
from auspex.exp_factory import QubitExpFactory
q4 = QubitFactory("q4")
seqs = [[X(q4),MEAS(q4)]]
filenames = compile_to_hardware(seqs, 'junk/junk')

exp = QubitExpFactory.create(filenames, expname="One-segment-issue", repeats=10) # repeats unnecessary
exp.run_sweeps()

Exception in thread Thread-9:
Traceback (most recent call last):
  File "C:\Users\qlab\Anaconda3\envs\py4auspex\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "c:\users\qlab\documents\github\auspex\src\auspex\plotting.py", line 78, in run
    self.status_sock.bind("tcp://*:%s" % self.status_port)
  File "zmq/backend/cython/socket.pyx", line 495, in zmq.backend.cython.socket.Socket.bind (zmq\backend\cython\socket.c:5653)
  File "zmq/backend/cython/checkrc.pxd", line 25, in zmq.backend.cython.checkrc._check_rc (zmq\backend\cython\socket.c:10014)
zmq.error.ZMQError: Address in use
```
How are we suppose to run one segment sweeps?!?

dieris, 2017-10-27:
The error you're getting is not because of the sweep (but a plot server problem). Have you tried restarting the kernel?

dieris, 2017-10-27:
But you're right, a single-point exp. won't save anything...

matthewware, 2017-10-27:
I'm trying out some fixes now and definitely think this is a bug not a feature.

grahamrow, 2017-11-08:
From further discussion with @matthewware, I think this was not an issue with the number of segments but the number of averages being 1. There may be a lingering issue with a singleton number of segments, but I think this should be okay given that spectroscopy works with many averages.

PR #172 (2017-11-10): Fix for single average. 
Average filter now acts like a passthrough filter for averaging axes of length 1. Should resolve #158.
commit 428610fb (2017-11-10): Average filter now acts like a passthrough filter for averaging axes of length 1. Should resolve #158
closed 2017-11-10 by grahamrow