BBN-Q/Auspex #160: Multiple sources for a WriteToHDF5

matthewware, 2017-10-29:
Not currently possible.  There's some issue with the stripping the connection names when there are multiple edges going to the same vertex:

https://github.com/BBN-Q/Auspex/blob/e3a71d56807f4fb22af518a6fb823c81034b6e9e/src/auspex/exp_factory.py#L358

Expected behavior would be for all the data to be packed under separate 'qX-main' groups in the h5 file.

grahamrow, 2017-10-30:
Current scheme is to use multiple writers with the same filename. There are a number of pitfalls when writing multiple streams to file asynchronously that would make the resulting filter code very complex. The h5 files are kept by the experiment class and rolled out to any and all filters that want to use them.

matthewware, 2017-10-30:
That seems reasonable.  I'll write up some doc on that.  Long term we might want to think about a way to make it easier as to set that up as the number of qubits goes up.  Not an immediate need though.

closed 2019-01-28 by grahamrow