BBN-Q/Auspex #182: Global dummy mode: expected behavior

matthewware, 2017-11-13:
Seems like `auspex_dummy_mode` only affects the Alazar, X6 and APS units.  Is this the behavior we should assume or should Auspex not try to connect to anything in 'dummy mode'?  I guess I'd vote for the later unless there's a strong reason for limiting this to sequencers and digitizers.

Could we just add a check in the base instrument class and use MagicMock if `auspex_dummy_mode` is true?  @grahamrow @gribeill @dieris what do you think?

grahamrow, 2017-11-13:
The main reason for dummy mode was to avoid hanging on to DLLs for the USB instruments (as you enumerated above) when all we wanted to do was scrape the auspex package with quince to create the nodes. Quince would hang on to the DLLs otherwise and cause problems.

I never had any intent beyond that — do you think there's a use case beyond the scraping?

matthewware, 2017-11-14:
The use case I had in mind was debugging when a fridge is warm and I don't want to actually connect to any instruments.  I could always nuke the yaml files or disable all the instruments.  It might not be useful to anyone else but I was thinking it would function as a kind of debug mode.

matthewware, 2019-04-12:
At the moment, we still need to link the idea of not loading the libraries and creating fake data.  Might also be nice to have an option to ignore all plotting for unit test.