BBN-Q/Auspex #184: Quince opens an empty graph

matthewware, 2017-11-13:
When Quince is run in a notebook Auspex can't seem to load ruamel and no nodes are created for the graph.

>python C:\Users\qlab\Documents\GitHub\Quince\run-quince.py C:\...\...\Documents\Config\yaml\measure.yml
Note: using QGLCfgFile [C:\...\...\Documents\Github\QGL\QGL\config.json]
Could not load Auspex because or error 'No module named 'ruamel''
Failed to load Auspex with error 'No module named 'ruamel''. There will be no nodes.

Though this process seems to be working fine from the Anaconda prompt...? @gribeill I know you've seen this issue as well.

matthewware, 2017-11-13:
Seems like this might be a Quince issue when Quince tries to import Auspex.

matthewware, 2017-11-13:
Think I'm closing in on an answer.  Looks like `Popen` is calling the run-quince.py in the wrong environment...

matthewware, 2017-11-13:
So I can confirm the suprocess.Popen is using the root conda env to execute the run-quince file.

grahamrow, 2017-11-13:
Yeah it's hard to glean the user's intent here. On windows we can only choose the environment within a .bat file since windows ignores any shebangs. On posix systems it will respect the shebang so you should change the shebang in your installed run-quince.py.

Any reason you're using multiple environments at this point? The main reason was that pyqt5 wasn't initially supported in pyqlab.

matthewware, 2017-11-14:
I use multiple envs but I guess we could hack this.  It forces the user to always run auspex and quince in their root envs.  There should be a way to activate the right Conda env before executing the batch file.  I think I'm close.

matthewware, 2017-11-14:
On second thought, this also assumes people are using conda so maybe the simplest thing would be to leave it as is and warn the user so they know what's going on.

grahamrow, 2017-11-14:
Agreed. I don't think it's unreasonable that the various BBN packages can only be expected to operate with one another if they're in the same env. If the user has multiple environments, I think that constitutes a special case.

I'm curious as to why quince launches for you at all. If you haven't installed it in the current anaconda environment it must be pulling the script from the root env? At any rate, we should probably do some checking to make sure quince() properly launches by examining the subprocess output for errors.

grahamrow, 2018-09-14:
Declaring this antiquated issue related to environments, as we move away from Quince.

closed 2018-09-14 by grahamrow