BBN-Q/Auspex #212: Fix multiprocessing DataBuffer implementation

grahamrow, 2018-01-10:
Previously the `DataBuffer.get_data()` method simply returned the buffers that had been accumulated over a data run. Now, for the multiprocessing branch, this present issues since the buffers are modified in a separate `Process` and need to be subsequently available back in the main thread.

**Possible implementations**
1. Shared `Array` with numpy views on either end
2. Output `Queue` in which the buffers are placed at the end of a run.

The latter is much easier in terms of data typing, at least.

dellard, 2018-01-17:
I've modified `DataBuffer` to place its final output into a `Queue` so that it can be accessed by the caller, and modified most of the tests to use this new interface (although not all of the other code, because of the problem described below.

Fixing `get_data()` is only part of the problem, unfortunately (at least, as far as the unit tests are concerned).  There are other members of `DataBuffer` that we try to access directly, and get garbage values, i.e. `finished_processing`.

If we want to do this cleanly, we need to wrap the Process objects in a "local" class that knows how to communicate with the "remote" Process instance.  A quicker method is to replace flags like `finished_processing` with `Event`s and rewrite accesses to them.  I'm still figuring out what callers might want to access.

grahamrow, 2018-09-14:
This is fixed, certainly as of #280

closed 2018-09-14 by grahamrow