BBN-Q/Auspex #213: Extend H5Handler Process to provide interface for returning intermediate data

grahamrow, 2018-01-11:
For adaptive sweeps, refine functions have typically accessed currently open H5 files in order to extract data and determine what new points the sweeper should use. In the multiprocessing conversion this is more difficult since the file objects are handled in separate `H5Handler` processes whose memory is not directly accessible.

We should extend the interface of the `H5Handler` in order that it can return data using an additional multiprocessing `Queue`. For example in `test_adapt_1d.py` we currently have:

```python
def test_writehdf5_1D_adaptive_sweep(self):
        exp = SweptTestExperiment()
        if os.path.exists("test_writehdf5_1D_adaptive-0000.h5"):
            os.remove("test_writehdf5_1D_adaptive-0000.h5")
        wr = WriteToHDF5("test_writehdf5_1D_adaptive.h5")

        edges = [(exp.resistance, wr.sink)]
        exp.set_graph(edges)

        def rf(sweep_axis, exp):
            logger.debug("Running refinement function.")
            temps = wr.group['data']['temperature'][:]
            ress  = wr.group['data']['resistance'][:]
```
in this case we should replace the latter lines with, e.g., `temps = wr.get_array('data/temperature')`. This will probably involve making the `WriteToHDF5` filter aware of its associated `H5Handler` around https://github.com/BBN-Q/Auspex/blob/feature/multiprocessing/src/auspex/experiment.py#L430

matthewware, 2019-02-15:
Functionality might already be in the new develop.

grahamrow, 2019-03-01:
Moot point, h5 gone. Intermediate data can be fetched at any rate.

closed 2019-03-01 by grahamrow