BBN-Q/Auspex #222: Are these configs still used?

matthewware, 2018-02-07:
Do all the configs come from the BBN_MEAS_FILE now?  Are these necessary:

https://github.com/BBN-Q/Auspex/blob/92317c172338455e7fd06d294d60b49349f10080/src/auspex/config.py#L35

??
@gribeill @grahamrow @dieris

grahamrow, 2018-02-08:
I guess part of the reason to keep them around was being able to programmatically set them without loading a text config file. The use case for that behavior is mostly things like unit testing, but I actually prefer the idea of loading test `.yml` files per unit test to better encapsulate them. Everyone else in favor of ditching them?

matthewware, 2018-02-08:
I'm okay with leaving them as long as we have a comment so new users don't get confused with the BBN_MEAS_FILE and other parameters in the config.

closed 2019-01-28 by grahamrow