BBN-Q/Auspex #227: failing unit tests cause failures in future unit tests

dellard, 2018-02-20:
I noticed that once some of the unit tests fail, they keep failing (even if the code is "fixed"), apparently because they leave H5 files (possibly corrupt) that interfere with later runs of the same tests.  There is no fix except to hunt down and remove the offending files.

I haven't tracked down whether this is a problem with the unit tests themselves, or whether the routines being tested are failing.  It's also not clear whether this is due to something not being initialized properly, or not being cleaned up properly (or both).

grahamrow, 2018-03-15:
Some tests have been better than others about cleaning up after themselves, but I've tried to make the tests better about cleaning up *before* they start running. This has mostly been on the multiprocessing branch.

grahamrow, 2019-03-01:
Now using temporary files and directories in any tests that write to file. Using in memory channel libraries also makes this much easier. Closing this until there are any other explicit issues.

closed 2019-03-01 by grahamrow