BBN-Q/Auspex #235: Add strip as additional_args in SCPICommand

appwagner, 2018-03-29:
I'm writing a driver for the Tektronix AWG5014 which obnoxiously adds \n to the beginning of its response strings. This prevents the transparent implementation of SCPICommands for the instrument. Can we add strip as an additional argument in SCPICommands so that strip() is automatically run on the responses from the instrument?

grahamrow, 2018-03-30:
You could try monkeypatching in the __init__ method of the instrument?
```python
self.interface.query = lambda s: self.interface.query(s).strip()
```
If that doesn't work we can add something more official

grahamrow, 2018-09-14:
Or, alternatively, is this actually an unread character from previous commands terminating in \r\n instead of just \r? Sorry I haven't merged this yet.

closed 2021-04-14 by gribeill