BBN-Q/Auspex #255: Sweep is much slower than qlab

jaseung, 2018-05-23:
I noticed that a sweep is much slower in Auspex than Qlab, at least a factor of 3 in the test I did. We have two Auspex setups in our lab, and we see the same issues on both setups. For a simple 1D scan, it might be still Ok, but when we want to take more data points like in a 2D scan, the sweep could be too slow to make Auspex undesirable in practice. Do you also see the same issue on your side(I am wondering if there is something we are doing wrong here)?

Let me describe below what I did for testing.
I just ran a cavity frequency scan with 51 number of data points in both Qlab and Auspex, and measured the time it takes to run sweep measurement (only sweep, not initial config time). I got,
-sweep in qlab  = 20 sec
-sweep in Auspex  = 63 sec

The repetition rate for the measurement was 50us, and I averaged 1000 points. For the filters in Auspex, I used only Stream-q1, avg-stream-q1 and Plot-Stream-q1 to minimize the number of nodes. For reference,  I attached the yml files that I used.

[Config-auspex.zip](https://github.com/BBN-Q/Auspex/files/2031489/Config-auspex.zip)

grahamrow, 2018-05-23:
Some computers (on windows in particular) are not particular speedy in assembling the filter pipeline and we've had to add a delay at line 127 of `exp_factory.py` for the sake of reliability that would account for 38s of your delay — more or less bringing auspex and qlab into parity.
```
await asyncio.sleep(0.75)
```
I would encourage seeing if you can reduce this delay. Part of the problem is that the `run` method is called for each cavity frequency — I might recommend baking some portion of the frequency sweep into your pulse sequence by means of modulating the pulse frequencies.

In general, sweeping hardware parameters outside of the pulse sequencer will always be slow.

jaseung, 2018-05-23:
Thanks for the explanation. 
I just tried 0.1 and 0.01 for sleep time, and the sweeps seemed to work fine. It took 32 and 27sec, respectively. If the sweep is reliable with 0.01, I might stick to 0.01.

closed 2018-05-23 by jaseung

grahamrow, 2018-05-23:
Good to know that this worked out!