BBN-Q/Auspex #257: Digitizer timeout (Alazar) during Rabi Chevron

jaseung, 2018-05-29:
During Rabi Chevron measurement (Rabi width vs qubit frequency), the digitizer (ATS9870) times out if the number of sweep points is larger, in this particular case, than 7. The measurement with a smaller sweep points works fine. I am testing with APS1+Alazar, but APS2+Alazar also has the same issue. Would you take a look to confirm and fix the issue?

[image]

jaseung, 2018-05-29:
No, it doesn't. I tried sleep(0.75) and sleep(2), but none worked. You haven't seen the same problem, at least with X6?

dieris, 2018-05-29:
Not a time out. I only saw some out-of-sync issue https://github.com/BBN-Q/Auspex/issues/120 which was resolved by adding some of that delay.

matthewware, 2019-04-12:
Don't know if this is still a major issue but I would also advise against doing any experiment with a length 0.  You're just asking for a variety of headaches.  I typically do something like np.linspace(0.02,0.52,51)*1e-6.  Are you still seeing this issue?  I imagine this is also a QGL issue related to the pulseBlock issues from a ~year ago https://github.com/BBN-Q/QGL/issues/148#issue-288969548.

jaseung, 2019-04-12:
Well, I don't know if this issue still remains in our setup. With the most current version of Auspex and QGL, there might not be such issue any more. We're not using Auspex ever since we tried to set Auspex up and encountered several issues about a year ago. Also, I am not asking for a variety of headaches, but just reporting them, hoping to get them fixed. Anyway, don't worry about this issue.

closed 2019-04-12 by jaseung

matthewware, 2019-04-12:
Thanks jaseung!