BBN-Q/Auspex #324: Kernel in database for X6 integrated streams

dieris, 2019-03-11:
Currently, the way to set a kernel in the bbndb branch is manually and through a file:

`cl['X6'].ch('integrated-1-1').kernel = np.loadtxt(os.path.join(config.KernelDir, 'q1-ss_kernel.txt'), dtype=complex, converters={0: lambda s: complex(s.decode().replace('+-', '-'))})`

We would like instead to save the kernel to database when running a `SingleShotFidelityExperiment` (that would require auspex/bbndb finding its way from the `SingleShotMeasurement` filter (or `FidelityKernel` node in the pipeline) to the appropriate integrated `ReceiverChannel`.

grahamrow, 2019-03-25:
Hmm, maybe let's defer this until we come up with the intended usage pattern for stream selectors?

gribeill, 2019-07-24:
BBN-Q/bbndb#16 lets the user load a kernel from a file to a stream selector, which resolves this a little.