BBN-Q/Auspex #335: Crash after few hours taking data

dieris, 2019-03-23:
To be investigated. Sorry for the dumb dump: 
[image]

grahamrow, 2019-03-25:
Hmm, you get it with ZMQ, I get it with multiprocessing. I'm testing out explicit garbage collection, and think we need to be extra careful to close things...

grahamrow, 2019-04-09:
There are now a bunch of changes on develop. Please advise on status when possible.

matthewware, 2019-04-12:
Highly likely this was due to the file descriptors issue Graham has been fixing.  @dieris is there something you could run over the weekend for a stress test?

dieris, 2019-04-12:
will try as soon as I switch back, on a different OS right now...

closed 2019-08-23 by dieris